SilverThings/bulldog #34: Refactor native parts of Bulldog Library

sbunciak, 2015-05-07:
Requirements: 
* use *only* memory access to control gpio pins. See PoC here: https://github.com/marvec/MemoryMap/
* keep support for all the following platforms: Raspberry Pi, CubieBoard, BeagleBoneBlack.

marvec, 2015-08-14:
Please note that some platforms provide their own ways of controlling pins. For example BBB ships with a kernel module and Bulldog library uses this module to control the pins via file system in /proc. This is very slow and it would be great to write to the shared memory directly. For that to be possible, the kernel module must be unloaded first.

sbunciak, 2015-11-04:
We could introduce some general implementation abstraction that will enforce usage of /dev/mem approach and the extending sub-classes would just provide memory configuration.

sbunciak, 2016-01-08:
Depends on #68