SilverThings/bulldog #40: FileNotFoundException

timmolter, 2015-07-09:
When running your example code using Maven, I get:

    java.io.FileNotFoundException: File /lib/bulldog-linux-native-beagleboneblack-hardfp.so was not found in classpath.

What else am I missing? I followed the instructions under `Usage`.

timmolter, 2015-07-09:
It looks like Maven isn't pulling `bulldog-linux-native-beagleboneblack-0.1.2.so` down from Maven Central.

timmolter, 2015-07-09:
I tried another approach to get this to work. I clone the repo and imported into Eclipse. When I run my LED blinking app, I get the following. Any help you can offer?

```
[main] ERROR io.silverspoon.bulldog.linux.sysinfo.CpuInfo - Cannot read file /proc/cpuinfo
java.nio.file.NoSuchFileException: /proc/cpuinfo
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:86)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
	at sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:214)
	at java.nio.file.Files.newByteChannel(Files.java:361)
	at java.nio.file.Files.newByteChannel(Files.java:407)
	at java.nio.file.Files.readAllBytes(Files.java:3152)
	at io.silverspoon.bulldog.linux.sysinfo.CpuInfo.readFile(Unknown Source)
	at io.silverspoon.bulldog.linux.sysinfo.CpuInfo.<clinit>(Unknown Source)
	at io.silverspoon.bulldog.beagleboneblack.BeagleBoneBlackBoardFactory.isCompatibleWithPlatform(BeagleBoneBlackBoardFactory.java:12)
	at io.silverspoon.bulldog.core.platform.Platform.createBoard(Unknown Source)
	at org.knowm.sbc_apps.bbb.BulldogLED.main(BulldogLED.java:14)
Exception in thread "main" java.lang.NullPointerException
	at io.silverspoon.bulldog.beagleboneblack.BeagleBoneBlackBoardFactory.isCompatibleWithPlatform(BeagleBoneBlackBoardFactory.java:12)
	at io.silverspoon.bulldog.core.platform.Platform.createBoard(Unknown Source)
	at org.knowm.sbc_apps.bbb.BulldogLED.main(BulldogLED.java:14)
```

sbunciak, 2015-07-11:
Can you reproduce with the 0.2.0-SNAPSHOT version?

timmolter, 2015-07-11:
I'll let you know eventually. I've moved on to other things for now.

commit 85aa46fe (2015-08-26): Removed arch from LinuxLibraryLoader (Fixes #40)

The arch was present because of name of library,
which was changed during refactoring.
closed 2015-09-04 by xjaros1