SilverThings/bulldog #46: Verify SPI issue reported in original Bulldog repository

sbunciak, 2015-08-08:
https://github.com/Datenheld/Bulldog/issues/18

xjaros1, 2016-01-14:
It looks that problem might be bigger,
currently bulldog fails on
```
Java HotSpot(TM) Client VM warning: You have loaded library /tmp/bulldog-linux-native9221056374032777446.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xab77b13c, pid=27040, tid=3057140832
#
# JRE version: Java(TM) SE Runtime Environment (8.0_60-b27) (build 1.8.0_60-b27)
# Java VM: Java HotSpot(TM) Client VM (25.60-b23 mixed mode linux-arm )
# Problematic frame:
# C  [bulldog-linux-native9221056374032777446.so+0x613c]  is_high+0x48
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /root/hs_err_pid27040.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted
```
So further investigation will be needed.

xjaros1, 2016-01-15:
Issue is unreproducable.

closed 2016-01-15 by xjaros1