SilverThings/bulldog #49: JVM crash on RPi B+

matoushybl, 2015-08-23:
Hi, I've been trying to use this library on my RaspberryPi B+ with Gradle and Maven, but in both cases the JVM crashed because of SIGSEGV. I have also tried using java 8 and 7 but with no success either. Can you please tell me which logs should I provide to help fixing this or is the problem somewhere in my configuration?

Here is the crash:
```#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xa8a9d228, pid=2676, tid=2830103664
#
# JRE version: Java(TM) SE Runtime Environment (7.0_40-b43) (build 1.7.0_40-b43)
# Java VM: Java HotSpot(TM) Client VM (24.0-b56 mixed mode linux-arm )
# Problematic frame:
# C  [bulldog-linux-native8866466634207559465.so+0x6228]  Java_io_silverspoon_bulldog_linux_jni_NativeMmap_getIntValueAt+0x0
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/javapi/hs_err_pid2676.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
#
Aborted
```
pom.xml
```
<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0"
         xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
         xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
    <modelVersion>4.0.0</modelVersion>

    <groupId>com.matoushybl</groupId>
    <artifactId>javapi</artifactId>
    <version>1.0-SNAPSHOT</version>

    <dependencies>
        <dependency>
            <groupId>io.silverspoon</groupId>
            <artifactId>bulldog-board-raspberrypi</artifactId>
            <version>0.1.2</version>
        </dependency>
    </dependencies>

    <build>
        <plugins>
            <plugin>
                <groupId>org.codehaus.mojo</groupId>
                <artifactId>exec-maven-plugin</artifactId>
                <version>1.3.2</version>
                <configuration>
                    <mainClass>com.matoushybl.javapi.Main</mainClass>
                </configuration>
            </plugin>
        </plugins>
    </build>
</project>
```

Main.java
```
package com.matoushybl.javapi;

import io.silverspoon.bulldog.core.gpio.DigitalOutput;
import io.silverspoon.bulldog.core.platform.Board;
import io.silverspoon.bulldog.core.platform.Platform;
import io.silverspoon.bulldog.raspberrypi.RaspiNames;

/**
 * @author <a href="mailto:hyblmatous@gmail.com">Matous Hybl</a>
 */
public class Main {

    public static void main(String[] args) {
        Board board = Platform.createBoard();

        DigitalOutput out = board.getPin(RaspiNames.P1_11).as(DigitalOutput.class);

        out.blinkTimes(2000, 500);
    }
}
```

Thanks

xjaros1, 2015-08-24:
Hello, RaspberryPi v2 is unsupported in version 0.1.2
If you want to use bulldog, please change version to 0.2.0-SNAPSHOT. The support for RaspberryPi 2 is implemented and will be released in version 0.2.0.

xjaros1, 2015-08-24:
Have you started your application under root user?

matoushybl, 2015-08-24:
The root permissions seem to solve it, but how come the lack of permission can make JVM crash?

commit 4485c619 (2015-09-04): Merge pull request #50 from px3/fix49

Added checks to memory mapping (fixes #49)
commit 78f044a1 (2015-09-04): Added checks to memory mapping (fixes #49)

This fix prevents JVM from crashing. When opening file or mapping memory
fails RuntimeException is thrown.
closed 2015-09-04 by xjaros1