SilverThings/bulldog #55: How to read digital input signal

max90727, 2015-10-30:
Hi. I write some code for reading the digital input, but I got the error as "could not create pin". Please help me.
                Board board = Platform.createBoard();
		DigitalInput digitalInput=board.getPin(BBBNames.P8_12).as(DigitalInput.class);
		Signal signal=digitalInput.read();
		System.out.println(signal.getNumericValue());

xjaros1, 2015-10-30:
Hello, are you running your code as root? Are there any exceptions printed?

pmacik, 2015-11-02:
Hi max90727, I fixed the code that most likely caused your issue. Can you please try your code now and confirm it works for you?

Thanks!

max90727, 2015-11-02:
Thanks for your help. My project goes fine now, and I found that I can read the signal from my temperature sensor by building the device tree. Anyway, still thanks for your help and I will try that when I have more time.

closed 2015-11-02 by max90727