SilverThings/bulldog #59: Required documentation on io.silverspoon.bulldog.raspberrypi.RaspiNames constants.

yangboz, 2015-11-05:
For example, Which name was matched on "Respberry Pi Model B+ V1.2" ?

xjaros1, 2015-11-05:
I don't understand the question. In RaspiNames are constants for pins only. Detection of platform is done automatically. Device layout is same for RaspberryPi A, B+ and 2, so there is no need for platform specific names if that's what you are asking for.

yangboz, 2015-11-05:
Let me re-clarify the question. According to example code at https://github.com/px3/bulldog
'''
//Set up a digital output
    DigitalOutput output = board.getPin(RaspiNames.P1_11).as(DigitalOutput.class);
'''
My question is: as a beginner I don't know  which constant parameter is suitable for  "board.getPin()", while I am using "Respberry Pi Model B+ V1.2" board?

Thanks,
Yangbo

xjaros1, 2015-11-05:
Ok that doesn't depend on board but on pin you want to use. I would recommend you to read this: 

http://elinux.org/RPi_Low-level_peripherals

Here are described pins on Raspberry PI and what are capabilities of those pins.

xjaros1, 2015-11-07:
I'm closing this issue.

closed 2015-11-07 by xjaros1