SilverThings/bulldog #78: PlexusIoResourceCollection not found when assembling plugin with dependencies into jar

benedekh, 2016-02-13:
Dear All,

I am trying to assemble my project with its dependencies into a single jar on my PC (both Windows 8.1 and Ubuntu 14.04). I added the next XML snippet into the pom.xml between `<plugin></plugin>` tags: 
    
    <artifactId>maven-assembly-plugin</artifactId>
    <executions>
        <execution>
            <phase>package</phase>
            <goals>
                <goal>single</goal>
            </goals>
        </execution>
    </executions>
    <configuration>
        <descriptorRefs>
            <descriptorRef>jar-with-dependencies</descriptorRef>
        </descriptorRefs>
    </configuration>
    
But I receive a long stack trace as a response, which says:

> Failed to execute goal org.apache.maven.plugins:maven-assembly-plugin:2.2-beta-5:single (default) on project bbb: Failed to create assembly: Error adding file-set for 'io.silverspoon:bulldog-linux-native-beagleboneblack:so:0.2.1' to archive: Error adding archived file-set. PlexusIoResourceCollection not found for: C:\Users\benjo\.m2\repository\io\silverspoon\bulldog-linux-native-beagleboneblack\0.2.1\bulldog-linux-native-beagleboneblack-0.2.1.so: No such archiver: 'so'. -> [Help 1]

You can find the full stack trace attached to this issue: [stacktrace.txt](https://github.com/px3/bulldog/files/129065/stacktrace.txt)

May anyone know what causes the problem?

Thank you.

xjaros1, 2016-02-13:
Hello, 
the problem is that maven assembly dooesn't know what to do with .so files, which are native parts of bulldog library.

To get this fixed just add to bulldog dependency exclusion, here is snippet

``` xml
 <dependencies>
	.
	.
	.
        <dependency>
            <groupId>io.silverspoon</groupId>
            <artifactId>bulldog-board-beagleboneblack</artifactId>
            <version>0.2.1</version> <!-- or version you would like to
use-->
            <exclusions>
                <exclusion>
                    <groupId>io.silverspoon</groupId>
                    <artifactId>bulldog-linux-native-beagleboneblack</artifactId>
                </exclusion>
            </exclusions>
        </dependency>
	.
	.
	.
    </dependencies>
```

This exclusion will cause, that maven assembly won't try to extract .so file.

benedekh, 2016-02-13:
Thank you. When I deploy the *-jar-with-dependencies.jar shall I copy anything else with it? E.g. the .so file.

xjaros1, 2016-02-13:
No, If you'll look inside the jar, you'll see that .so files are already in there, this just prevents from opening the .so file.

xjaros1, 2016-02-17:
I suppose, that your question was answered, I'm closing this issue.

closed 2016-02-17 by xjaros1