SilverThings/bulldog #87: error with digital input in cubieboard

jcarvalhogo, 2016-03-07:
Greetings friends, I'm loving bulldog however I am with the following problem:
Connecting to 192.168.1.40:22
cmd : cd '/home/cubie/NetBeansProjects//TesteLed'; '/usr/lib/jvm/java-8-oracle/jre/bin/java'  -Dfile.encoding=UTF-8   -jar /home/cubie/NetBeansProjects//TesteLed/dist/TesteLed.jar 
/dev/mem: Permission denied
Exception in thread "main" 
io.silverspoon.bulldog.linux.util.MMapFailedException: Unable to open file /dev/mem
at io.silverspoon.bulldog.linux.io.mmap.MemoryMap.<init>(MemoryMap.java:34)
at io.silverspoon.bulldog.linux.io.mmap.MemoryMap.<init>(MemoryMap.java:28)
at io.silverspoon.bulldog.cubieboard.gpio.CubieboardGpioMemory.<init>(Unknown Source)
at io.silverspoon.bulldog.cubieboard.Cubieboard.<init>(Unknown Source)
at io.silverspoon.bulldog.cubieboard.Cubieboard.getInstance(Unknown Source)
at io.silverspoon.bulldog.cubieboard.CubieboardBoardFactory.createBoard(Unknown Source)
at io.silverspoon.bulldog.core.platform.Platform.createBoard(Platform.java:26)
at br.com.Led.main(Led.java:26)
Exception in thread "Thread-0" 
java.lang.NullPointerException
	at io.silverspoon.bulldog.cubieboard.Cubieboard.cleanup(Unknown Source)
	at io.silverspoon.bulldog.core.platform.AbstractBoard$1.run(AbstractBoard.java:136)
/Users/Josemar/Dropbox/Clientes/TesteLed/nbproject/remote-platform-impl.xml:105: The following error occurred while executing this line:
/Users/Josemar/Dropbox/Clientes/TesteLed/nbproject/remote-platform-impl.xml:63: Remote command failed with exit status 1
BUILD FAILED (total time: 18 seconds)

I just can not run with root after I run twice I have get message:
IOException: java.nio.file.NoSuchFileException: /sys/class/gpio/gpio8/direction

my GPIO this directory like this:
ls /sys/class/gpio/
export  gpio8_pg2  gpiochip1  unexport
cubie@Cubian:~/NetBeansProjects/TesteLed/dist$

I believe the error is here:  gpio8_pg2 is != gpio8 
I tried to compile the sources with the possible solution however was unable

I was doing something to type below
public class Gpio {

    public static final byte PG03 = 1;
    public static final byte PB19 = 2;
    public static final byte PB18 = 3;
    public static final byte PG06 = 4;
    public static final byte PG05 = 5;
    public static final byte PG04 = 6;
    public static final byte PG01 = 7;
    public static final byte PG02 = 8;
    public static final byte PG00 = 9;
    public static final byte PH14 = 10;
    public static final byte PH15 = 11;
    public static final byte PI06 = 12;
    public static final byte PI05 = 13;
    public static final byte PI04 = 14;
    public static final byte PG11 = 15;
    public static final byte PG10 = 16;
    public static final byte PG09 = 17;
    public static final byte PG08 = 18;
    public static final byte PG07 = 19;
    public static final byte PE08 = 20;
    public static final byte PE07 = 21;
    public static final byte PE06 = 22;
    public static final byte PE05 = 23;
    public static final byte PE04 = 24;
    public static final byte PI09 = 25;
    public static final byte PI08 = 26;
    public static final byte PI07 = 27;
    public static final byte PD04 = 28;
    public static final byte PD03 = 29;
    public static final byte PD02 = 30;
    public static final byte PD01 = 31;
    public static final byte PD00 = 32;
    public static final byte PE11 = 33;
    public static final byte PE10 = 34;
    public static final byte PE09 = 35;
    public static final byte PD12 = 36;
    public static final byte PD11 = 37;
    public static final byte PD10 = 38;
    public static final byte PD09 = 39;
    public static final byte PD08 = 40;
    public static final byte PD07 = 41;
    public static final byte PD06 = 42;
    public static final byte PD05 = 43;
    public static final byte PD20 = 44;
    public static final byte PD19 = 45;
    public static final byte PD18 = 46;
    public static final byte PD17 = 47;
    public static final byte PD16 = 48;
    public static final byte PD15 = 49;
    public static final byte PD14 = 50;
    public static final byte PD13 = 51;
    public static final byte PB02 = 52;
    public static final byte PD25 = 53;
    public static final byte PD24 = 54;
    public static final byte PD26 = 55;
    public static final byte PD27 = 56;
    public static final byte PD23 = 57;
    public static final byte PD22 = 58;
    public static final byte PD21 = 59;
    public static final byte PI11 = 60;
    public static final byte PI13 = 61;
    public static final byte PI10 = 62;
    public static final byte PI12 = 63;
    public static final byte PB13 = 64;
    public static final byte PB11 = 65;
    public static final byte PB10 = 66;
    public static final byte PH07 = 67;

    private static Map<Integer, String> gpios = null;

    public static void initGpioMap() {
        gpios = new HashMap<>();
        gpios.put(1, "gpio1_pg3");
        gpios.put(2, "gpio2_pb19");
        gpios.put(3, "gpio3_pb18");
        gpios.put(4, "gpio4_pg6");
        gpios.put(5, "gpio5_pg5");
        gpios.put(6, "gpio6_pg4");
        gpios.put(7, "gpio7_pg1");
        gpios.put(8, "gpio8_pg2");
        gpios.put(9, "gpio9_pg0");
        gpios.put(10, "gpio10_ph14");
        gpios.put(11, "gpio11_ph15");
        gpios.put(12, "gpio12_pi6");
        gpios.put(13, "gpio13_pi5");
        gpios.put(14, "gpio14_pi4");
        gpios.put(15, "gpio15_pg11");
        gpios.put(16, "gpio16_pg10");
        gpios.put(17, "gpio17_pg9");
        gpios.put(18, "gpio18_pg8");
        gpios.put(19, "gpio19_pg7");
        gpios.put(20, "gpio20_pe8");
        gpios.put(21, "gpio21_pe7");
        gpios.put(22, "gpio22_pe6");
        gpios.put(23, "gpio23_pe5");
        gpios.put(24, "gpio24_pe4");
        gpios.put(25, "gpio25_pi9");
        gpios.put(26, "gpio26_pi8");
        gpios.put(27, "gpio27_pi7");
        gpios.put(28, "gpio28_pd4");
        gpios.put(29, "gpio29_pd3");
        gpios.put(30, "gpio30_pd2");
        gpios.put(31, "gpio31_pd1");
        gpios.put(32, "gpio32_pd0");
        gpios.put(33, "gpio33_pe11");
        gpios.put(34, "gpio34_pe10");
        gpios.put(35, "gpio35_pe9");
        gpios.put(36, "gpio36_pd12");
        gpios.put(37, "gpio37_pd11");
        gpios.put(38, "gpio38_pd10");
        gpios.put(39, "gpio39_pd9");
        gpios.put(40, "gpio40_pd8");
        gpios.put(41, "gpio41_pd7");
        gpios.put(42, "gpio42_pd6");
        gpios.put(43, "gpio43_pd5");
        gpios.put(44, "gpio44_pd20");
        gpios.put(45, "gpio45_pd19");
        gpios.put(46, "gpio46_pd18");
        gpios.put(47, "gpio47_pd17");
        gpios.put(48, "gpio48_pd16");
        gpios.put(49, "gpio49_pd15");
        gpios.put(50, "gpio50_pd14");
        gpios.put(51, "gpio51_pd13");
        gpios.put(52, "gpio52_pb2");
        gpios.put(53, "gpio53_pd25");
        gpios.put(54, "gpio54_pd24");
        gpios.put(55, "gpio55_pd26");
        gpios.put(56, "gpio56_pd27");
        gpios.put(57, "gpio57_pd23");
        gpios.put(58, "gpio58_pd22");
        gpios.put(59, "gpio59_pd21");
        gpios.put(60, "gpio60_pi11");
        gpios.put(61, "gpio61_pi13");
        gpios.put(62, "gpio62_pi10");
        gpios.put(63, "gpio63_pi12");
        gpios.put(64, "gpio64_pb13");
        gpios.put(65, "gpio65_pb11");
        gpios.put(66, "gpio66_pb10");
        gpios.put(67, "gpio67_p7");
    }

    public static String getGpio(byte pin) {
        return gpios.get(pin & 0xFF);
    }

}

can anybody help me?

xjaros1, 2016-03-07:
Hello,

the problem is, that bulldog doesn't use /sys to access gpio, it uses /dev/mem instead. You need to run you app with root permisions in order to access /dev/mem. This is due to direct access to registers.

Best Regards

jcarvalhogo, 2016-03-07:
right, thank you for your reply!
lease you could describe an example using digital input? I intend to count the pulse encoder.

I'm using this code to test the digital input, and it is with him to give the error

//Detect the board we are running on
        Board board = Platform.createBoard();

        //Set up a digital output
        //DigitalOutput output = board.getPin(CubieboardNames.PG1).as(DigitalOutput.class);
        DigitalInput input = board.getPin(CubieboardNames.PG2).as(DigitalInput.class);

        System.out.println("Foi bem assim mesmo ... " + input.read());

thank you!!

jcarvalhogo, 2016-03-07:
Hello, still can not use Digital Input on cubieboard, below the test code.

public static void main(String[] args) {
     //Grab the platform the application is running on
    	Board board = Platform.createBoard();
    	
    	//Set up a digital input
    	DigitalInput buttonSignal = board.getPin(CubieboardNames.PG1).as(DigitalInput.class);
    		
    	//Create the button with this DigitalInput
    	Button button = new Button(buttonSignal, Signal.High);
    	
    	//Add a button listener
    	button.addListener(new ButtonListener() {

                        @Override
			public void buttonPressed() {
				System.out.println("PRESSED");
			}

                        @Override
			public void buttonReleased() {
				System.out.println("RELEASED");
			}
    		
    	});
    	
    	while(true) {
    		BulldogUtil.sleepMs(50);
    	}
   
    }

someone could tell me where I am going wrong?

this below is the content of my folder dist
README.TXT  
TesteLed.jar  
bulldog-linux-native-cubieboard.so  
lib

below this the log of execution:
root@Cubian:/home/cubie/NetBeansProjects/TesteLed/dist# java -jar TesteLed.jar 
a4901000, a4901000
IOException: java.nio.file.NoSuchFileException: /sys/class/gpio/gpio7/direction

my operating system and Cubian X1 nano

Help me prease !!!