SilverThings/bulldog #88: Build Breaking

ajesk, 2016-04-06:
Hey, I have recently stumbled upon your library for Beaglebone/Java implementation and see that your repository is more up to date than libbulldog. I was wondering if I could get some assistance in building the library. Currently I have followed the instructions for putting everything together and ran the command to compile bulldog for arm. The build is failing after a number of tests on the following process.
[INFO] io.silverspoon:bulldog-linux-native-cubieboard .... FAILURE [0.089s]

Could someone shed some light on why the build might be breaking at this point and possibly help guild the completion.

Thanks,
A

mijaros, 2016-04-06:
Can you tell me how are you building the library? Do you have installed gcc arm cross compiler and are you properly invoking in maven?

ajesk, 2016-04-06:
I thought I did as I had run 'apt-get gcc-4.8' and it was already installed. but found that I did not have gcc-4.8-arm-linux-gnueabihf. After downloading that I am still getting the same error. I also have gcc-linaro-5.2-2015.11-2-x86_64_arm-linux-gnueabihf sitting in my home directory (I required it to compile kernels previously). 

Is there a specific way which I need to invoke them? or am I completely off in how I am approaching this?

Thanks,
A

ajesk, 2016-04-06:
And I apologize the command that I attempted to use was.

mvn clean install -Pnative-compile

mijaros, 2016-04-06:
Well no, if you'll try to read Readme you'll find this 

> mvn clean install -Dcompiler.exec=arm-linux-gnueabihf-gcc-4.8 -Dlinker.exec=arm-linux-gnueabihf-ld

Please try this, it should resolve your issue.

ajesk, 2016-04-06:
That did the trick!!

Thank you!

closed 2016-04-06 by mijaros