SilverThings/bulldog #89: Issue with loading libraries

ajesk, 2016-04-07:
I apologize for continuing to open up issues on this, but there is little outlet for assistance or documentation to aid in the use of this library collection.

This time I am being presented with the following stack after attempting to run my program on my beagleboard (which has now be slightly altered to use bulldog instead of the retired libbulldog):
Exception in thread "main" java.lang.UnsatisfiedLinkError: no bulldog-linux in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1864)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.bulldog.linux.util.LinuxLibraryLoader.loadNativeLibrary(LinuxLibraryLoader.java:14)
	at org.bulldog.beagleboneblack.BeagleBoneBlackBoardFactory.createBoard(BeagleBoneBlackBoardFactory.java:16)
	at org.bulldog.core.platform.Platform.createBoard(Platform.java:13)
	at org.bulldog.core.platform.Platform$createBoard.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:112)
	at eissbox.control.beagle.BeagleControlPoint.<init>(BeagleControlPoint.groovy:93)

It is occurring when I am attempting to create the board from within my system (Platform.createBoard()). From looking at the source it seems like it is having an issue finding the .so files. I have placed them in a number of locations as well as altered the -Djava.library.path to attempt to include these files, but to no avail.

Here is how I am running my application:
java -Xmx30m -cp "lib/*" my.Program

Thank you,
Aaron

ajesk, 2016-04-08:
I I have solved this issue by means that I am not sure of.

closed 2016-04-08 by ajesk

cjritola, 2018-10-08:
This is still an issue. Getting it from a .jar built with Maven Assembly. Version 0.3.0 and 0.3.1-SNAPSHOT same result. BeagleBoneBlack running Jessie. the JAR has the .so file in its lib/ directory.

ajesk, 2018-10-13:
If I recall correctly, I remember changing the permissions of the .so files, and copying them to another root level dir in order to get it to work. I wish I could give more insight, but time has tarnished my memory of all of the nonsense I went through to get this to work.

MaxMbems, 2018-12-03:
@cjritola  Did you ever find a step by step solution to this? I've tried the above which hasn't seemed to work

cjritola, 2019-01-17:
Didn't find a solution. Right now just using a workaround with a small C program which communicates with the java part via stdout/stdin.