SilverThings/bulldog #96: Fix BeagleBoardBlack error "java: free(): invalid next size (fast): " when use interrupt handler (Button listener)

igor-suhorukov, 2016-09-27:
When program use Interrupt handler on BeagleBoardBlack native code exception is occured

Java program:
        String pinName = args[0];
        Board board = Platform.createBoard();
        DigitalInput isClose = board.getPin(pinName).as(DigitalInput.class);

        Button button = new Button(isClose, io.silverspoon.bulldog.core.Signal.Low);
        button.addListener(new ButtonListener() {

            @Override
            public void buttonPressed() {
                System.out.println("PRESSED");
            }

            @Override
            public void buttonReleased() {
                System.out.println("RELEASED");
            }
        });

        for(int i=0; i<100; i++){
            System.out.println(isClose.read().getNumericValue());
            Thread.sleep(300);
        }

        Thread.sleep(20000);
        System.out.println("DONE");

-------------
machinekit@beaglebone:~$ uname -a
Linux beaglebone 3.8.13-xenomai-r79 #1 Wed Jun 15 20:40:46 UTC 2016 armv7l GNU/Linux

---------
OUTPUT...

0
0
0

*** glibc detected *** /home/machinekit/jdk1.8.0_101//bin/java: free(): invalid next size (fast): 0x001a2d58 ***

igor-suhorukov, 2016-10-17:
The same error on debian BeagleBoard.org Debian Image 2015-03-01
---
uname -a
Linux beaglebone 3.8.13-bone70 #1 SMP Fri Jan 23 02:15:42 UTC 2015 armv7l GNU/Linux
---
root@beaglebone:/home/debian# gpio-bbb-1.0-SNAPSHOT/bin/launcher.sh P9_18
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
-- listing properties --
beagleboneblack.emmc.enabled=true
beagleboneblack.hdmi.enabled=true
beagleboneblack.dogtag=BeagleBoard.org Debian Image 2015-03-01

iolib_setdir: PortSet_ptr P9.18
0
*** glibc detected *** /home/debian/jdk1.8.0_101//bin/java: malloc(): memory corruption (fast): 0x001e30b8 ***
Aborted

igor-suhorukov, 2016-10-17:
*** glibc detected *** /home/debian/jdk1.8.0_101//bin/java: munmap_chunk(): invalid pointer: 0x00269b68 ***
Aborted

ZenoFuturista, 2016-11-02:
Hello all,
I faced the same issue on my Beaglebone black and found a workaround, so I would like to let you know of my findings. Versions of my software:
uname -a
Linux alarm 4.8.5-1-ARCH #1 Sat Oct 29 20:34:37 MDT 2016 armv7l GNU/Linux

java -version
openjdk version "1.8.0_112"
OpenJDK Runtime Environment (build 1.8.0_112-b15)
OpenJDK Zero VM (build 25.112-b15, interpreted mode)

So, at the beggining I was using crosscompiled version of Bulldog from my desktop PC (arm gcc 6.1.1) and I was facing similar errors when attaching interrupt listeners to Input GPIOS. From what I saw, the problems came from Bulldog natives. Afterwards I tried and recompiled Bulldog on the Beaglebone itself (gcc 6.2.1) and the problem went away. Could it be that there is some mismatch in headers? 
I hope this could point you out to successfull solution. 
Have a nice days.
Jan